Klimadatastyrelsen/AutoBernese #50: Byg BSW-API container

xidus, 2023-02-22:


kbevers, 2023-02-22:
Hvad betyder det her egentligt? Er det noget med Docker?

xidus, 2023-02-22:
`gensta`:

* Der skal være adgang til ftp-serveren hos EUREF for at man kan hente STA-filen.
* [x] Jeg har bestilt adgang til tre FTP-servere, som er nævnt i den aktive autobpe.conf-fil, som bruges på MWs server.

xidus, 2023-05-01:
*   Opret kampagne
    -   Givet:
        +   GPS-uge
    -   Opret en ny standard-kampagne, der kan vælges i Bernese GUI'en.
        +   Er kampagnen ikke en standard-kampagne, kan man altid lave en basis-kampagne i GUI'en og tilføje en kampagne-konfiguration.

xidus, 2023-05-04:
Status på denne er, at der nu kan oprettes en mappe med kampagnen i en undermappe af Bernese-kampagne-mappen, og her kopieres en skabelon over, som brugeren selv skal lave. Som udgangspunkt kommer der en default skabelon med, når man opretter kampagner. Mappen med skabeloner skal vedligeholdes af brugerne og ligger ,hvor autobernese har sine 'runtime' filer så som log-filen og andet. default.skabelonen kommer med pakken og overføres, hvis ikek mappen med skabeloner findes i forvejen. Ellers forbliver mappen urørt.

Næste opgave er at få tilføjet de nødvendige filer fra Bernese-installationen til den oprettede kampagne, samt oprette filtrætet fra kampagne-mappens rod.

Når dette er gjort, kan man oprette en kampagne i Bernese med autobernese.

Næste issue handler om, hvordan man henter data til kampagnen, og dernæst igen skal BPE-processerne håndteres.

xidus, 2023-05-22:
Man kan nu oprette en kampagne baseret på kampagne-navn (mappenavn i roden af standard-bernese kampagne-mappe), skabelon-fil, start- og slut-dato (begge inklusive i beregningen) for kampagnen.

closed 2023-05-22 by xidus