Klimadatastyrelsen/AutoBernese #54: Logging

larsnaesbye, 2023-02-25:
Vi talte om at logging på et generelt plan er noget vi gerne vil implementere tidligt, fordi det allerede i udviklingsfasen er et nyttigt værktøj.

Som den relativt mindre Python-erfarne vil jeg så spørge om det indbyggede [`logging`](https://docs.python.org/3/library/logging.html) modul kan udfylde vores behov tilstrækkeligt, eller om vi bør bruge et andet.

larsnaesbye, 2023-02-25:
Lukker pga. #55 er blevet konverteret til issue

closed 2023-02-25 by larsnaesbye

kbevers, 2023-02-26:
> Som den relativt mindre Python-erfarne vil jeg så spørge om det indbyggede [logging](https://docs.python.org/3/library/logging.html) modul kan udfylde vores behov tilstrækkeligt, eller om vi bør bruge et andet.

`logging` er det oplagte valg. De fleste tredjeparts pakker bruger det også hvilket gør det ret nemt at trække logging info fra fx SQLALchemy ind i sine egne logs. Jeg bookmarkede for noget tid siden denne artikel som jeg synes giver en fin rundtur i logging med Python:

https://guicommits.com/how-to-log-in-python-like-a-pro/