Klimadatastyrelsen/AutoBernese #64: Måske genoverveje `ab` som executable navn

larsnaesbye, 2023-03-02:
Der findes allerede en `ab` præinstalleret på mange servere/computere, nemlig [ApacheBenchmark](https://twiki.org/cgi-bin/view/Codev/ApacheBenchmark) som er en executable på `/usr/sbin/ab`.

larsnaesbye, 2023-03-03:
Ellers skal vi nok lige lave en beskrivelse af hvordan man vælger den rigtige. For Apache Benchmark kommer præinstalleret med moderne macOS. 

`ab` er ellers dejligt hurtigt at skrive på en kommandolinje. Jeg kan ikke finde noget der hedder `aube`, det næst-korteste.

kbevers, 2023-03-03:
Vores udgangspunkt er at køre programmet på en ubuntu linux, så vi behøver ikke tage de store hensyn til hvad der følger med en Mac (kan man overhovedet bruge Bernese på MacOS?). Men når det er sagt, så snart man aktiverer ab-miljøet er vores ab-kommando den der dukker op. I hvert fald her på min Mac.

larsnaesbye, 2023-03-03:
Man kan godt compile og installere Bernese på macOS (nogen har f.eks. installeret 5.2 [her](https://gilles.ecgs.lu/installing-bernese-5-2-on-macos/)).

Jeg har åbenbart problemer selv efter aktivering af conda-miljøet:

```
(base) lanch@Lars-M1 AutoBernese % conda activate ab-dev                  
(ab-dev) lanch@Lars-M1 AutoBernese % ab
ab: wrong number of arguments
Usage: ab [options] [http[s]://]hostname[:port]/path
Options are:
    -n requests     Number of requests to perform
    -c concurrency  Number of multiple requests to make at a time
    -t timelimit    Seconds to max. to spend on benchmarking
                    This implies -n 50000
    -s timeout      Seconds to max. wait for each response
 ```

kbevers, 2023-03-04:
> Man kan godt compile og installere Bernese på macOS (nogen har f.eks. installeret 5.2 [her](https://gilles.ecgs.lu/installing-bernese-5-2-on-macos/)).

Det er jo også bare Fortran så i virkeligheden så overraskende at det kan lade sig gøre, men det er nok temmelig usandsynligt at vi kommer til at gøre det. Trods alt.

> ...og selv om pakken er installeret:

Jeg fik det samme i første omgang, men efter at have deaktiveret og aktiveret miljø'et står vores `ab` først i PATH. I hvert fald hos mig.

Jeg er hverken for eller imod at ændre navnet. Jeg tror bare ikke det er så stort et problem i praksis. Især i det systemlandskab vi skal arbejde.

larsnaesbye, 2023-03-04:
> Jeg fik det samme i første omgang, men efter at have deaktiveret og aktiveret miljø'et står vores `ab` først i PATH. I hvert fald hos mig.

Ah ja! Deaktiver og genaktiver virker også her. Jeg lukker issuet.

closed 2023-03-04 by larsnaesbye