Klimadatastyrelsen/AutoBernese #97: Create Campaign fejler

nho-sdfe, 2023-06-13:
(ab) e004243@gnss:/home/bsw/dev-nho$ ab campaign create -b 2019-10-22 -e 2019-10-24 WEEK2076
Traceback (most recent call last):
  File "/home/e004243/mambaforge/envs/ab/bin/ab", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/ab/cli.py", line 286, in create
    _campaign.create(name, template, beg, end)
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/site-packages/ab/bsw/campaign.py", line 233, in create
    shutil.copy(fname_source, path_full / Path(fname_source).name)
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/shutil.py", line 419, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/home/e004243/mambaforge/envs/ab/lib/python3.11/shutil.py", line 256, in copyfile
    with open(src, 'rb') as fsrc:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/bsw/dev-nho/BERN54/GLOBAL/CONFIG/OBSERV.SEL'

(ab) e004243@gnss:/home/bsw/dev-nho/data/CAMPAIGN54/WEEK2076$ ls
ATM  BPE  GEN

xidus, 2023-06-20:
Tak. Det ser ud til, at den direkte årsag er, at filen ikke er downloaded endnu.

Løsningen er, at du først downloader de generelle kilder med `ab download`-kommandoen. Denne leder efter sektionen `sources` i den samlede konfiguration, så her skal du oprette tekstfilen `/home/bsw/dev-nho/autobernese/autobernese.yaml` og som minimum indsætte følgende indhold:

```yaml
sources:

- !Source
  name: BSW Configuration data
  url: ftp://ftp.aiub.unibe.ch/BSWUSER54/CONFIG/
  destination: *CONFIG
  filenames: ['*']
  max_age: 1
```

Når du har gjort dette, så kan du opdatere filerne i $CONFIG-mappen med dem fra FTP-serveren.

Herefter burde du kunne oprette kampagnen uden problemer (jeg testede det i dev-miljøet, hvor kilderne er downloadede).

**Årsagen** er, at dén `.SEL`-fil, der kommer med installationspakken hedder noget andet end dén, vi er enige om at bruge og som altså kan hentes på FTP-serveren. Dette er den eneste årsag til, at du ikke kan oprette en kampagne uden først at opdatere disse filer (og dermed downloade den manglende fil `OBSERV.SEL`).

Vil du se resten af de kilder, vi har defineret, kan du se dem i filen `/home/bsw/dev/autobernese/autobernese.yaml` :)

kbevers, 2023-06-20:
@xidus er der en fidus i at fejle på pænere måde, evt med en besked om at der mangler opsætning i stil med det du beskriver herover?

xidus, 2023-06-20:
Jo, det er meningen, at vi med brugertests får en afprøvning af brugergrænsefladen, jeg som udvikler ikke oplever, fordi jeg kender applikationen.

Fejlhåndteringen skal laves i `cli` modulet, og der er plads til validering både her, samt i de Python-typer, jeg har lavet YAML-tags til, og hvor noget af valideringen også kunne være.

De konkrete tilfælde, jeg selv er vidne til hos brugerne, vil jeg skrive ned i GitHub issues, så de kan komme i kø til at blive arbejdet på (eller afvist).

xidus, 2023-06-21:
Jeg lukker denne, da vi gennemgik løsningen sammen igår.

closed 2023-06-21 by xidus